javmarina/Nintendo-Switch-Remote-Control #1: How to build Java parts?

Sch1nken, 2020-04-24:
Hey there, I've stumbled upon your great work while trying to build my own solution with the original reverse engineered code. As I am basically trying to do the same thing you did, I tried building yours.

I tried "gradle build" from the gui folder. I got some jar files but no executable jars. I've only ever worked with gradle in the context of android studio and generally don't have much java experience. Can you help me?

Thank in advance.

javmarina, 2020-04-26:
I'm using IntelliJ IDEA, so you have to configure artifacts. There is a nice official guide [here](https://www.jetbrains.com/help/idea/working-with-artifacts.html). When configured, you will need to build with Build > Build artifacts menu.

There are two artifacts: one for the client and another one for the server. That way you can generate two jars.

Note that you can also Run/Debug the client and server modules.

javmarina, 2021-03-22:
Hi @Sch1nken, I have created a new Gradle task called `buildAll` that will generate the server and client JARs. Just run `gradlew buildAll` and you're good to go. If you don't need to make changes to the code, you can use the [precompiled JARs](https://github.com/javmarina/Nintendo-Switch-Remote-Control/releases/tag/latest). Tell me if it works!

closed 2021-11-02 by javmarina