javmarina/Nintendo-Switch-Remote-Control #2: Issue after flashing in DFU mode for Arduino Uno r3

SequentialBase9915, 2020-06-03:
I successfully flashed the ATmega16u thanks to your directions and the Arduino is recognized correctly as a controller. However I’ve found that in both this build and that found in SwitchInputEmulator, I cannot get the Switch to receive commands.

I’ve done the wiring you described in your message to the SwitchInputEmulator dev and still no luck, were there any other modifications you made to get the setup to work?

I will add that I have an FT232 adaptor and Arduino Uno r3, and can confirm the setup works as I can send commands to the Switch successfully via the build in the Switch-Fightstick repository.

This is all amazing work and thanks in advance!

javmarina, 2020-06-03:
You said the board is recognized as a controller. Does that mean that the Switch shows it as a USB controller in the menu? You should see that and you can also try connecting it to the PC, when it has to appear as "HORIPAD S" or something like that.
If that is the case, it means you flashed the firmware correctly, so the issue is about communications. I didn't make any relevant change in respect of SwitchInputEmulator.
Does your FT232 adaptor has LEDs? In that case, you can check if the commands are being sent. You should see both RX and TX leds on, because the board responds with an ACK message after every corerctly received packet (or NACK).
If you use the Java program, the connection status is shown in the GUI. If the Arduino doesn't recognize packets correctly, you will see the message "Syncing..." very often, because the host will try to resync after some NACKs.
Finally, the error might be the baudrate. The baudrate is programmed in the firmware and my code has it set to 1Mbps. If you select a different baudrate in the Java program, it won't work.

Tell me if some of those things work!

SequentialBase9915, 2020-06-10:
Thanks a lot for the feedback, that was a good place to start and appeared to in fact to be a part of the problem. I adjusted the baud rate to match on the serial/USB adapter and made sure the serial port initialization matched those values and did get it to recognize on a separate testing laptop.

I found that the order of connecting the USB Serial, Arduino and Switch was important. I had to disconnect all three parts, first plug in the Arduino USB to Switch, then plug in USB Serial adapter to PC and finally wire the adapter to the Arduino (TX->TX RX->RX GND->GND 5V->5V) led to consistent successful connectivity. After that I had to make sure the sync protocol ran (I used the one from SwitchInputEmulator which initialized at 19200 baud) and it worked as needed.

I hope this can maybe help anyone else struggling to connect and thanks for your help!

SequentialBase9915, 2020-07-29:
Apologies I hadn’t had the time to test but I tried the steps you mentioned and those worked fine as well, thank you.

javmarina, 2020-07-29:
Ok, so I will leave the readme as it is now. Thanks!

closed 2020-07-29 by javmarina