javmarina/Nintendo-Switch-Remote-Control #3: Controller at index 0 is not connected!

Tails32, 2021-03-20:
The Java Client component sadly does not work for me. It seems to be a problem with the Jamepad library, but since I can't really pin the issue down, I share this here for now. I tested under Windows 10 20H2, Feature Experience Pack 120.2212.1070.0 with a Xbox Series controller, a DualShock 4 controller and the virtual Xbox 360 controller from Parsec, all of them causing the same issue. To make sure I also used another machine entirely, also Windows 10. I also tried letting some else build the JAR for me, since I am not really sure if I did it correctly, but that didn't help either. Using the client with a keyboard works fine, I can control the Switch with that, but this doesn't cover my use case for this project.

```
NATIVE METHOD: loading from  "C:\Users\tails\AppData\Local\Temp\6964659601243710076.tmp"
             result 64  Failed loading libusb-1.0.dll: The specified module could not be found.

2 service(s) available
 0. Keyboard
com.studiohartman.jamepad.ControllerUnpluggedException: Controller at index 0 is not connected!
        at com.studiohartman.jamepad.ControllerIndex.ensureConnected(ControllerIndex.java:271)
        at com.studiohartman.jamepad.ControllerIndex.getName(ControllerIndex.java:250)
        at com.javmarina.client.services.DefaultJamepadService.toString(DefaultJamepadService.java:90)
        at com.javmarina.client.Client.showInitialFrame(Client.java:80)
        at com.javmarina.client.Client.main(Client.java:54)
 1. Disconnected
```

javmarina, 2021-03-20:
That definitely doesn't seem like a controller issue, but you can double check it (search them in the Device Manager).
I recall having some issues with Steam and a Switch Pro Controller. If you have Steam installed and running, try force closing it and check again.
If that doesn't work, try removing line 171 in Client.java (`controllers.quitSDLGamepad();`) and rebuild the client artifact. I don't think it will work, but you have nothing to loose.
Finally, it could be a library issue (although I hope it's not). I'm using the latest version of Jamepad (1.4.0). Regarding sdl2gdx, I have wanted to use it for quite some time now, but it seems like I would have to refactor a big part of the client code.

Btw, how did you come across this repository? I would like to know how this project can make it to more people :)

Tails32, 2021-03-21:
> That definitely doesn't seem like a controller issue, but you can double check it (search them in the Device Manager).

The controllers definitely work. All three of them that I've tested. I can see them in Windows's Game Controllers Settings and all inputs work in the Test tab. I could only make out the Xbox controller in the device manager though. It was labeled as "XINPUT-compatible HID device".

> I recall having some issues with Steam and a Switch Pro Controller. If you have Steam installed and running, try force closing it and check again.

I've tried that too before opening this issue, just forgot to mention this. Sorry. Wouldn't want to bother someone without trying everything I can do. I have Steam installed and have it setup for managing the controller input in games, but even closing it and checking in Task Manager for remnants of steam.exe or alike sadly didn't change a thing.

> If that doesn't work, try removing line 171 in Client.java (controllers.quitSDLGamepad();) and rebuild the client artifact. I don't think it will work, but you have nothing to loose.

This suggestion actually helped a little. I still can't seem to send any controller inputs to the server, but at least the list now shows some kind of descriptive name for the controllers I'm using and after connecting it doesn't instantly shutdown both client and server anymore.
They are labeled as follows:
- Xbox Series Controller => RAWINPUT:0x045e/0x02ff
- DualShock 4 Controller => HIDAPI:PS4 Controller
- (can't test Parsec's virtual Xbox 360 right now)

After connecting though, I get the following exception:
```
Exception in thread "Thread-0" java.lang.IllegalStateException: SDL is already initialized!
	at com.studiohartman.jamepad.ControllerManager.initSDLGamepad(ControllerManager.java:71)
	at com.javmarina.client.JamepadManager.addService(JamepadManager.java:24)
	at com.javmarina.client.services.DefaultJamepadService.onStart(DefaultJamepadService.java:79)
	at com.javmarina.client.services.ControllerService.start(ControllerService.java:78)
	at com.javmarina.client.Client$StartButtonListener.lambda$actionPerformed$0(Client.java:320)
	at java.base/java.lang.Thread.run(Thread.java:834)
```

>Finally, it could be a library issue (although I hope it's not). I'm using the latest version of Jamepad (1.4.0). Regarding sdl2gdx, I have wanted to use it for quite some time now, but it seems like I would have to refactor a big part of the client code.

I've already tried using sdl2gdx but since this is pretty much new ground for me I failed at it. It was an amateurish try and I failed miserably. Also tried changing out the version of Jamepad used here for the one in libgdx/Jamepad but that didn't work out as well. But these tries are nothing to rely on. If you want to look into this, better try it yourself. High chance I just failed somewhere.

> Btw, how did you come across this repository? I would like to know how this project can make it to more people :)

I came across this repository about two weeks ago. The idea to stream the Switch with Parsec or similar tools and then use it for online coop already came to my mind a few times before that, but it was almost impossible for me to find a solution with Google since as soon as you combine the terms "PC", "Switch" and "Controller", all it gives you are tutorials to use the Pro Controller on PC games. I was lucky and came across a video from Parsec's YouTube channel which showed off using a "Titan One" to control the Switch from a PC, but their products seemed to me far too expensive. Therefore I searched once again and finally (after testing various search terms) found your repository with "switch controller emulator". Your solution looked perfectly tailored for my needs and the hardware didn't seem too expensive or hard to get. Also, I could reuse the Arduino and the USB to Serial Adapter for other things too whenever I want to. Then I ordered all the parts and began trying everything out. At first I had issues with flashing the firmware under Windows. Later I used Ubuntu to try the Linux flash tool dfu-programmer. Then I noticed that my USB to Serial Adapter didn't connect properly (I got one with a CP2102 chip first), then ordered another one since it was cheap and had the mentioned CH340. That helped me to connect the server program to the board. Now I'm here where we are now. Struggling with literally the last hurdle. Controller input. I'm so close to getting it to work...

Lastly, thank you so much for taking your time to look into my issue. I feared that you might be too busy or maybe were not interested in this project anymore since the last relevant seeming commit I found was in mid August, but it seems that these worries were baseless. Maybe we can find a solution to this together. Thank you very much! 😄

javmarina, 2021-03-21:
> I could only make out the Xbox controller in the device manager though. It was labeled as "XINPUT-compatible HID device".

That's fine, controllers sure work correctly.

> This suggestion actually helped a little. I still can't seem to send any controller inputs to the server, but at least the list now shows some kind of descriptive name for the controllers I'm using and after connecting it doesn't instantly shutdown both client and server anymore.

Well, that's a big step. My code worked fine in my computer, but maybe I was doing something wrong. To be more specific, I initialize the SDL manager to fetch available controllers, then quit the manager and, only if the user selects a Jamepad controller, initialize it again. It seems that the SDL manager can't be used again after it is closed. So my suggestion is that you keep that change and also modify `addService()` in `JamepadManager` like this:

    public static void addService(final DefaultJamepadService service) {
        pool.add(service);
    }

So you just have to remove the `if` statement. This is a temporary solution, just to see if it works. If so, I would make more robust changes to the code.

> If you want to look into this, better try it yourself

The thing is, it works fine for me! I really don't know what's happening.

> Lastly, thank you so much for taking your time to look into my issue. I feared that you might be too busy or maybe were not interested in this project anymore since the last relevant seeming commit I found was in mid August, but it seems that these worries were baseless.

I'm still really interested in this project, but other priorities in my life mean I can't spend as much time as I would want with this project. I have some experimental changes yet to commit, but nothing stable. Anyways, it made me very happy to hear from someone trying to use my code, even though it seems like nothing is working as expected 😅.

**Last thing**: if my suggestion works, can you test it with and without Steam running? I want to confirm if Steam running in the background is a real issue, so I can put it in the README. Thanks in advance.

Tails32, 2021-03-21:
Thank you so much for your help. The suggested changes worked! But the key mapping for the DS4 was "wrong", so I made a DualShock4ControllerService class which maps the buttons according to Nintendo's layout. I also had to invert the y-axis there. To detect it, I tested for "PS4 Controller" in the controller's name just like you checked for "Nintendo". The Xbox Series controller did not work over USB, but over Bluetooth. When connected over Bluetooth it is shown as follows:

- Xbox Series (over Bluetooth) - XInput Controller

I guess that the mapping just has to be remade for the Xbox Series controller over USB. Over Bluetooth the y-axis was also inverted, but I won't try to fix that because "XInput Controller" is just too generic to change it's behavior. Also, over Bluetooth ABXY are mapped to ABXY, which means that the letters match, but the button layout is actually inverted. That could use some getting used to it.

If you want me to share my DS4 class or the other changes, I could create a pull request. But since you said this is only a temporary solution, I don't know how much sense this would make.

> Last thing: if my suggestion works, can you test it with and without Steam running? I want to confirm if Steam running in the background is a real issue, so I can put it in the README. Thanks in advance.

The controllers worked exactly the same, whether Steam was running or not. I don't think it is an issue in my case. But just to clarify: my controller configurator is enabled for both Xbox and PlayStation controllers. It does not seem to interfere in any way.

> > If you want to look into this, better try it yourself

> The thing is, it works fine for me! I really don't know what's happening.

I just wanted to say, don't rely on my foolish attempts regarding the libraries. I'm not experienced enough in that, so better check switching to sdl2gdx yourself if you want to try that. 😅 Would be maybe better to use sdl2gdx since it works by letting the user assign their own button inputs as far as I've read into that, but it also makes more work for you. Sorry, I don't think I can help there. 😔

> I'm still really interested in this project, but other priorities in my life mean I can't spend as much time as I would want with this project. I have some experimental changes yet to commit, but [...]

If you have other things to do, then just focus on those. Anyway, thanks again that you still take your time to help me out. I think your code worked fine. Most issues I had came from not having much experience working with IntelliJ and buying the wrong hardware. The server just worked and the client too, even if it was just with the keyboard. The only issue was controller input and that seems to get resolved now, so don't worry. 😃

javmarina, 2021-03-21:
First of all, regarding the button mapping, I have not tested any controller outside the Switch Pro Controller, if I'm honest. I guessed the mapping was wrong just for Nintendo controllers, because of the ABXY layout vs. Xbox XYAB layout. The button mapping is taken from https://github.com/gabomdq/SDL_GameControllerDB and I wonder if I can do something that works for most controllers without fine-tuning. Maybe a GUI for manually configuring the button mapping, but I think that might be going a little too far.

> If you want me to share my DS4 class or the other changes, I could create a pull request. But since you said this is only a temporary solution, I don't know how much sense this would make.

I'm going to try and apply the changes we talked about, test them and push to the repo (then I will close this issue). So you can safely create a pull request afterwards!

> The controllers worked exactly the same, whether Steam was running or not. I don't think it is an issue in my case. But just to clarify: my controller configurator is enabled for both Xbox and PlayStation controllers. It does not seem to interfere in any way.

That's good to know! I will leave the README as it is right now.

> Most issues I had came from not having much experience working with IntelliJ and buying the wrong hardware.

If you came across something that seemed difficult and underexplained, let me know! I can write more detailed guides.

javmarina, 2021-03-21:
Should be fixed by the [latest commit](https://github.com/javmarina/Nintendo-Switch-Remote-Control/commit/9a296c81dcd25177718078f531cdd15f009357d2). Please check if it works and I will close the issue. Also, feel free to create a pull request whenever you want 😀.

Tails32, 2021-03-21:
I've created my own fork for now, because I made some changes to fit my needs. You can find it here: https://github.com/Tails32/Nintendo-Switch-Remote-Control

If you find the changes fitting, you can use them as you like. With my changes, every controller works the same regarding the layout of the buttons. The right face button (circle or Xbox B) is now Nintendo's right face button (the A button) for example.

I've tested the changes with:

- Switch Pro Controller (USB & Bluetooth)
- Xbox Series Controller (Bluetooth)
- DualShock 4 Controller (USB & Bluetooth)
- Virtual Xbox 360 Controller (Parsec)

Only the USB connection of the Xbox Series Controller did not work at all. All of the above behave the same as described above.

closed 2021-03-21 by Tails32

javmarina, 2021-03-21:
Hi @Tails32 

I've tested the [second commit](https://github.com/Tails32/Nintendo-Switch-Remote-Control/commit/632d9577655599195f2e9c2edac1d7692a643ec5) and works fine with my DS4, so I guess what I did in NintendoControllerService.java is indeed needed for all controllers. Guess that's what happens when you only test with one controller 😅.

Regarding the [first commit](https://github.com/Tails32/Nintendo-Switch-Remote-Control/commit/aee6ed2962bbbff50892cfaccae9ca84289e524d), it works with the DS4, but I don't recall having issues with the Pro Controller. I can't test it right now, so you can create a pull request now if you want, and I will test the Pro Controller in two weeks.

Btw, you might want to take a look at my last commit.

Thank you so much for your help!

Tails32, 2021-03-22:
I have to thank you! With your help I got it to work and I can look forward to many online coop games with my friends. 😄 Also thank you so much for the last commit. This makes it so much easier to build the JARs. But I would still recommend to put up a release here on GitHub if the changes in #4 work for you too. It's much easier for those who never used make or gradle to set this up and use the project like this. But I leave the decision to you, I don't want to talk you into something you don't want.

On a side note; I wondered whether it would be possible to simulate multiple controllers with only one board, since the Switch kind of supports USB hubs. I imagine implementing this would be quite a bit of work, but if you think it could be possible (and would be interested to look into it at some point) I would open another issue as a feature request.

javmarina, 2021-03-22:
> But I would still recommend to put up a release here on GitHub

That's a great idea and I don't know how I never though about it. JARs are now compiled automatically for each new commit in the master branch and put in the _Releases_ tab, see the last commits for more details.

> On a side note; I wondered whether it would be possible to simulate multiple controllers with only one board, since the Switch kind of supports USB hubs.

I'm not sure if that's possible. I'm using the LUFA library and I don't know if it allows to emulate multiple USB devices from a single microcontroller. I will take a look into it, but for me it's not a priority.

Tails32, 2021-03-23:
Since you seemed to like the idea with the GitHub releases, here's one last suggestion: Could you also add the firmware file Joystick.hex? That would simplify the process to just connecting the boards and flashing the firmware.

javmarina, 2021-03-23:
That's a little trickier because the compiled firmware depends on the board you use (there is no single Joystick.hex). In fact, if you flash the wrong firmware, you could brick the microcontroller. I could publish multiple firmwares but I don't want users to inadvertently flash the wrong firmware and then blame me. I will think about it.