javmarina/Nintendo-Switch-Remote-Control #6: Python code gets stuck in loops.

MyHeroYoshi, 2021-12-24:
I have tried to get the Python code to run, but the program keeps freezing.

The first time was at line 517. After commenting it out and finding the correct port index, it stopped at line 474 wait_for_data(). The method was called by sync(), which calls force_sync().

But when I start the Python program, the Arduino Uno R3 is recognised as a controller.

It worked via client.jar and server.jar, but unfortunately I had a CPU load of 100% and so I wanted to test python.
I would be very happy if the switch could be controlled via a python program.

javmarina, 2021-12-24:
The CPU usage of 100% is strange. It's high due to video encoding/decoding, but I didn't expect it to be **so** high.
I will try to look into the issue, but keep in mind that with the provided Python code, you can control the Switch via a **local** PC, but not remotely. Remote play is only implemented in Java.

javmarina, 2021-12-24:
Also, what do you mean when you say "The first time was at line 517". The program is expected to stop so that you can see the serial devices available and select one.

javmarina, 2021-12-24:
I found a bug in the code, but it's unrelated. I can't reproduce the issue right now. Can you detail your setup? OS, Python and library versions, hardware, etc?

javmarina, 2021-12-26:
I think I have fixed the issue, let me know if the last version works @MyHeroYoshi

MyHeroYoshi, 2022-01-04:
@javmarina Thank you, I will test it in the next few days. Because of the holidays and New Year, I hadn't had it on my radar.

MyHeroYoshi, 2022-01-05:
Wonderful. Works without problems. 

I have one more question, which has nothing to do with this issue.
Do I understand correctly that a command (including no input) must always be sent to the switch? Otherwise you always end up in the home screen.

javmarina, 2022-01-06:
Yes, that's true. To be more precise, a command must always be sent to the Arduino board. The Switch periodically requests input from the Arduino, and it delivers the last command received from the serial port. 

This mechanism effectively pauses the game when no input is received for a moment, this is done so that if remote connection is lost, the game is paused. It can be disabled by modifying the firmware source code, recompiling and reflashing it.

closed 2022-01-06 by javmarina