javmarina/Nintendo-Switch-Remote-Control #7: Stuck at COMMAND_SYNC_1's response as 0 on Arduino Uno R3

Uniersr, 2022-03-29:
### Video with Visual Log:
https://user-images.githubusercontent.com/99698024/160506021-bcb9ae89-f223-4bca-8ae4-abfd1824bc00.mp4
### 4 Photos of 4 alligator clips (RX -> RX, TX -> TX, 5V (Since Arduino Uno R3 doesn't has VCC if i'm not mistaken) -> 5V, GND -> GND)
[image]
[image]
[image]
[image]
### Nintendo Switch's Version is 14.0.0 (This is probably be the reason why it doesn't recognize arduino uno r3 since the last compiled developement build is 26 Dec 2021 and 14.0.0 is 21 Mar 2022) (On Arduino Uno R3, L light flashes 2 or 3 times when reset button on arduino uno r3 then it's off until reset button is pressed again so L light flashes 2 or 3 times again...)
[image]
Text Log:
```
Bytes written
Received 9 bytes: FF FF FF FF FF FF FF FF FF
RESP_SYNC_START received as last byte
Sending COMMAND_SYNC_1
Response: 0
Couldn't sync
Exception in thread "JavaFX Application Thread" java.lang.NullPointerException: Cannot invoke "javafx.stage.Stage.close()" because "this.stage" is null
        at com.javmarina.server.ConnectionFrame.close(ConnectionFrame.java:215)
        at com.javmarina.server.ConnectionFrame.show(ConnectionFrame.java:75)
        at com.javmarina.server.Server.lambda$start$1(Server.java:65)
        at com.javmarina.server.ServerController.lambda$setButtonAction$10(ServerController.java:268)
        at com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:86)
        at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:238)
        at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:191)
        at com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(CompositeEventDispatcher.java:59)
        at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
        at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
        at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
        at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
        at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
        at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
        at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
        at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:49)
        at javafx.event.Event.fireEvent(Event.java:198)
        at javafx.scene.Node.fireEvent(Node.java:8890)
        at javafx.scene.control.Button.fire(Button.java:203)
        at com.sun.javafx.scene.control.behavior.ButtonBehavior.mouseReleased(ButtonBehavior.java:206)
        at com.sun.javafx.scene.control.inputmap.InputMap.handle(InputMap.java:274)
        at com.sun.javafx.event.CompositeEventHandler$NormalEventHandlerRecord.handleBubblingEvent(CompositeEventHandler.java:218)
        at com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:80)
        at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:238)
        at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:191)
        at com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(CompositeEventDispatcher.java:59)
        at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
        at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
        at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
        at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
        at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
        at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
        at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
        at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:54)
        at javafx.event.Event.fireEvent(Event.java:198)
        at javafx.scene.Scene$MouseHandler.process(Scene.java:3862)
        at javafx.scene.Scene.processMouseEvent(Scene.java:1849)
        at javafx.scene.Scene$ScenePeerListener.mouseEvent(Scene.java:2590)
        at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:409)
        at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:299)
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
        at com.sun.javafx.tk.quantum.GlassViewEventHandler.lambda$handleMouseEvent$2(GlassViewEventHandler.java:447)
        at com.sun.javafx.tk.quantum.QuantumToolkit.runWithoutRenderLock(QuantumToolkit.java:412)
        at com.sun.javafx.tk.quantum.GlassViewEventHandler.handleMouseEvent(GlassViewEventHandler.java:446)
        at com.sun.glass.ui.View.handleMouseEvent(View.java:556)
        at com.sun.glass.ui.View.notifyMouse(View.java:942)
        at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
        at com.sun.glass.ui.win.WinApplication.lambda$runLoop$3(WinApplication.java:174)
        at java.base/java.lang.Thread.run(Thread.java:833)

```

Uniersr, 2022-03-30:
Currently, i'm working around to bypass it, in that time, it's stuck at "No packets were sent correctly".

[image]

javmarina, 2022-04-01:
Hi! Sorry for the delay, I've had a busy week.

It's super strange that the response is displayed as 0. It's either a bug in the Java side, or an electrical failure (short circuit?), because the Arduino is not supposed to send 0x00 never.

I think the Switch version is not relevant, because the issue has to do with the PC <-> Arduino connection, not with Arduino <-> Switch. Afaik, nothing to modify in the FW regarding Switch version, only if they changed the Pro Controller protocol, which I doubt.

How did you bypass the syncing?? It's a key feature so that packets are interpreted correctly by the Arduino.

I will try to investigate the Java exception, but it doesn't seem critical.

Uniersr, 2022-04-03:
> How did you bypass the syncing?? It's a key feature so that packets are interpreted correctly by the Arduino.
Hmmmmm, so the syncing is still necessary tho.

I confirm that my arduino and USB-to-UART are probably working since the python communicate from USB-to-UART to arduino for it to turn the light on and off via "l" and "d" commands on python which they're configured on arduino side.

https://user-images.githubusercontent.com/99698024/161444148-dfe4eb12-c962-4181-9bc4-970e0f92ba76.mp4

Arduino side:
```
char cmd;
void setup() {
  Serial.begin(19200);         
  pinMode(LED_BUILTIN, OUTPUT);
}
void loop() {
 if (Serial.available() > 0) { 
  cmd = Serial.read();        
  if (cmd == 'l') {           
    digitalWrite(LED_BUILTIN, HIGH); 
    Serial.println("Okay, the light is on");
  }
  else if (cmd == 'd') {      
    digitalWrite(LED_BUILTIN, LOW);
    Serial.write("Okay, the light is off");
  }
 }
}
```

javmarina, 2022-04-03:
> Hmmmmm, so the syncing is still necessary tho.

Yes. If no syncing is performed, the messages could be interpreted incorrecly. Messages with the controller input are 8 bytes. If not synced, byte 1 could be read as byte 2, thus sending the Switch a wong input.

I have solved the Exception in the text log, should be fixed with the last commit.

Great setup! I have analyzed the code and discovered a potential issue with how bytes are read from the serial port. ~~Can you check that Arduino -> PC messages are also sent? In the video we can see that PC -> Arduino is working, but I don't know about the other direction~~ Sorry, according to the logs, the Java progam is in fact receiving FF bytes.

Please, also tell me the baudrate you are using, because it might have something to do with the issue.

My guess is that the issue lies in SerialAdapter.java:112. The readByte() method would return 0 (which appears in the log) if the read timeout is elapsed. In this case, serialPort is not null. My proposal is that you change READ_TIMEOUT to 0, if you're able to.

It seems that you are using Windows. Did you enable low latency mode? See [here](https://projectgus.com/2011/10/notes-on-ftdi-latency-with-arduino/), try changing it in the Windows Control Panel.

Uniersr, 2022-04-04:
I flashed your hex file and used server.jar from last [Development Build](https://github.com/javmarina/Nintendo-Switch-Remote-Control/releases/tag/latest) so i disconnected blue cable to my PC and connected it again on my pc, it recognizes as Pro Controller so then i disconnected and connected my blue cable to my nintendo switch then RX - RX, TX - TX from arduino to my PC and i found regression even if i try the baudrate combinations of 1000000 and 9600, it would simply not work...

[image]
[image]
[image]

Next, i modified READ_TIMEOUT to 0 from last commit and compiled using "gradle build", nothing has changed.
[image]
[image]

Then i modified WRITE_TIMEOUT and READ_TIMEOUT to 1000 again and used "gradle clean build" this time, and the last hope is definitely lost as it fails again for the third time.
[image]
[image]

So i'll wait for your next reply or for someone to test this with the same hardware i used: Geninue Original Arduino Uno R3 from official arduino.cc store and USB-to-UART model (Usb 2.0 P Rs232 Ttl Uart Cp2102) being recognized as Silicon Labs CP210x USB to UART Bridge (COM6)

nullstalgia, 2022-04-07:
Can you take a photo of the pinout of your UART-USB board?

Also, in your pictures here and in the issue in my repository, I do not see a wire properly connecting into GND of the UART board and the Arduino. 

[image]

Here is how my connection looks.

GND - GND
TX - TX
RX - RX (Normally these would be swapped, but this is an unusual use case for the hardware)

I didn't load HoodLoader2 when testing it just now. Low chance, but could you have a weird AVR fuse setup? I'm assuming you have a way to use the ISP, and using AVRDUDESS I got these fuses from my 16u2. HoodLoader2 uses different fuses.

[image]

Uniersr, 2022-04-08:
> Can you take a photo of the pinout of your UART-USB board?

Ok!
[image]

And Here's the pinout of my arduino! (RX, TX and GND)
[image]

javmarina, 2022-04-09:
> No.

Perfect!

> I flashed [Arduino_UNO_R3.hex](https://github.com/javmarina/Nintendo-Switch-Remote-Control/releases/download/latest/Arduino_UNO_R3.hex) from [Development Build](https://github.com/javmarina/Nintendo-Switch-Remote-Control/releases/tag/latest) to my arduino one more time then i disconnected my arduino to my pc then i inserted TX-TX, RX-RX, 5V-5V, GND-GND to my USB-to-UART then my PC then blue cable connected to my nintendo switch TX-TX, RX-RX doesn't turn on the TXD and RXD lights on USB-to-UART adapter and arduino after enter "python client.py 1000000" then "0" the sync fails after a few seconds, after swapping to TX-RX, RX-TX does turn on the TXD and RXD lights on USB-to-UART except arduino then turn off and still sync fail in under than a second, i think we'll wait for someone to test the same thing with geninue original arduino uno r3 from arduino.cc and USB-to-UART named "Usb 2.0 P Rs232 Ttl Uart Cp2102" which is not FTDI unfortunately to see if appears the same issue as mine. so do i need to install HoodLoader2 to my arduino?

I don't know what HoodLoader2 is and I didn't do anything with it. From what I've seen I don't think it will help us.

See last commit, I have added more messages to client.py (untested). Try again with that, if it doesn't work try increasing timeout variable in line 524. My bet is that there is some HW issue and the PC is never receiving bytes!! Either because the adapter is malfunctioning or the Arduino is not sending them at all.

Uniersr, 2022-04-11:
> See last commit, I have added more messages to client.py (untested). Try again with that

```
List of available ports:
[0] COM6
Introduce port index: 0
Writing ['0x0', '0x0', '0x8', '0x80', '0x80', '0x80', '0x80', '0x0', '0x54'] to serial port
Wrote 9 bytes out of 9
Reading 1 byte(s) from serial port
Read 1 bytes: ['0x0']
[sync()] Not synced, forcing sync...
First, write nine 0xFF bytes to flush out buffer on device
Writing ['0xff', '0xff', '0xff', '0xff', '0xff', '0xff', '0xff', '0xff', '0xff'] to serial port
Wrote 9 bytes out of 9
7 byte(s) in the input buffer, waiting 18.0 ms for more bytes
17 byte(s) in the input buffer, waiting 18.0 ms for more bytes
24 byte(s) in the input buffer, waiting 18.0 ms for more bytes
30 byte(s) in the input buffer, waiting 18.0 ms for more bytes
35 byte(s) in the input buffer, waiting 18.0 ms for more bytes
42 byte(s) in the input buffer, waiting 18.0 ms for more bytes
48 byte(s) in the input buffer, waiting 18.0 ms for more bytes
53 byte(s) in the input buffer, waiting 18.0 ms for more bytes
59 byte(s) in the input buffer, waiting 18.0 ms for more bytes
65 byte(s) in the input buffer, waiting 18.0 ms for more bytes
71 byte(s) in the input buffer, waiting 18.0 ms for more bytes
76 byte(s) in the input buffer, waiting 18.0 ms for more bytes
77 byte(s) in the input buffer, waiting 18.0 ms for more bytes
Eventually we received 77 byte(s)
[SYNC ERROR] We expected between 1 and 9 bytes
[sync()] Not synced
Could not sync!
Writing ['0x0', '0x4', '0x1', '0x80', '0x1', '0x27', '0xd9', '0x0', '0x8f'] to serial port
Wrote 9 bytes out of 9
Reading 1 byte(s) from serial port
Read 1 bytes: ['0x0']
Packet Error!
Writing ['0x0', '0x0', '0x8', '0x80', '0x80', '0x80', '0x80', '0x0', '0x54'] to serial port
Wrote 9 bytes out of 9
Reading 1 byte(s) from serial port
Read 1 bytes: ['0x0']
Packet Error!
```
>  if it doesn't work try increasing timeout variable in line 524

Setting 2 timeout variables to 5.
`timeout = 5`
`ser = serial.Serial(port=available_ports[index].device, baudrate=args.baudrate, timeout=5)`

```
List of available ports:
[0] COM6
Introduce port index: 0
Writing ['0x0', '0x0', '0x8', '0x80', '0x80', '0x80', '0x80', '0x0', '0x54'] to serial port
Wrote 9 bytes out of 9
Reading 1 byte(s) from serial port
Read 1 bytes: ['0x0']
[sync()] Not synced, forcing sync...
First, write nine 0xFF bytes to flush out buffer on device
Writing ['0xff', '0xff', '0xff', '0xff', '0xff', '0xff', '0xff', '0xff', '0xff'] to serial port
Wrote 9 bytes out of 9
10 byte(s) in the input buffer, waiting 5000.0 ms for more bytes
15 byte(s) in the input buffer, waiting 5000.0 ms for more bytes
20 byte(s) in the input buffer, waiting 5000.0 ms for more bytes
26 byte(s) in the input buffer, waiting 5000.0 ms for more bytes
34 byte(s) in the input buffer, waiting 5000.0 ms for more bytes
39 byte(s) in the input buffer, waiting 5000.0 ms for more bytes
44 byte(s) in the input buffer, waiting 5000.0 ms for more bytes
51 byte(s) in the input buffer, waiting 5000.0 ms for more bytes
56 byte(s) in the input buffer, waiting 5000.0 ms for more bytes
63 byte(s) in the input buffer, waiting 5000.0 ms for more bytes
68 byte(s) in the input buffer, waiting 5000.0 ms for more bytes
74 byte(s) in the input buffer, waiting 5000.0 ms for more bytes
77 byte(s) in the input buffer, waiting 5000.0 ms for more bytes
Eventually we received 77 byte(s)
[SYNC ERROR] We expected between 1 and 9 bytes
[sync()] Not synced
Could not sync!
Writing ['0x0', '0x4', '0x1', '0x80', '0x1', '0x27', '0xd9', '0x0', '0x8f'] to serial port
Wrote 9 bytes out of 9
Reading 1 byte(s) from serial port
Read 1 bytes: ['0x0']
Packet Error!
Writing ['0x0', '0x0', '0x8', '0x80', '0x80', '0x80', '0x80', '0x0', '0x54'] to serial port
Wrote 9 bytes out of 9
Reading 1 byte(s) from serial port
Read 1 bytes: ['0x0']
Packet Error!
```
[image]

javmarina, 2022-04-13:
Hi, I have pushed a new commit. The PC is receiving a lot of bytes, maybe they were received before and stored in the input buffer. Last commit should fix that. Also, now the received bytes are printed (for easier debugging) and I have fixed a very obvious issue with the `read_byte_latest()` function. 😅

Uniersr, 2022-04-13:
```
List of available ports:
[0] COM6
Introduce port index: 0
Writing ['0x0', '0x0', '0x8', '0x80', '0x80', '0x80', '0x80', '0x0', '0x54'] to serial port
Wrote 9 bytes out of 9
Reading 1 byte(s) from serial port
Read 1 bytes: ['0x0']
[sync()] Not synced, forcing sync...
First, clear input buffer
Then, write nine 0xFF bytes to flush out buffer on device
Writing ['0xff', '0xff', '0xff', '0xff', '0xff', '0xff', '0xff', '0xff', '0xff'] to serial port
Wrote 9 bytes out of 9
7 byte(s) in the input buffer, waiting 18.0 ms for more bytes
13 byte(s) in the input buffer, waiting 18.0 ms for more bytes
19 byte(s) in the input buffer, waiting 18.0 ms for more bytes
25 byte(s) in the input buffer, waiting 18.0 ms for more bytes
30 byte(s) in the input buffer, waiting 18.0 ms for more bytes
36 byte(s) in the input buffer, waiting 18.0 ms for more bytes
41 byte(s) in the input buffer, waiting 18.0 ms for more bytes
46 byte(s) in the input buffer, waiting 18.0 ms for more bytes
47 byte(s) in the input buffer, waiting 18.0 ms for more bytes
Eventually we received 47 byte(s)
[SYNC ERROR] We expected between 1 and 9 bytes
[sync()] Not synced
Could not sync!
Writing ['0x0', '0x4', '0x1', '0x80', '0x1', '0x27', '0xd9', '0x0', '0x8f'] to serial port
Wrote 9 bytes out of 9
Reading 1 byte(s) from serial port
Read 1 bytes: ['0x0']
Packet Error!
Writing ['0x0', '0x0', '0x8', '0x80', '0x80', '0x80', '0x80', '0x0', '0x54'] to serial port
Wrote 9 bytes out of 9
Reading 1 byte(s) from serial port
Read 1 bytes: ['0x0']
Packet Error!
```

javmarina, 2022-04-14:
My bad, try again with the latest commit. However, it's super strange that you're receiving 47 bytes! There is something going on with the Arduino firmware. With the last change we will be able to see what we are receiving. Is the Arduino connected to the Switch via USB? Try disconnecting it (just serial between Arduino and PC) and powering the Arduino with a 5V cable from the serial adapter.

ethanriverpage, 2022-04-14:
I have the same Arduino && same USB to TTL adapter.

I could not get the client.py to read anything from the server -- I would consistently get the error `Couldn't sync with the AVR MCU`. 

Poking around with drivers -- I've noticed that I can manually adjust the baudrate of the COM4 port in Device Manager. It would not work at the default 1M baudrate. I recompiled the `Joystick.hex`, adjusting `avr.h` to a baudrate of 9600. I then adjusted the baudrate within the driver to 9600.

Works! Issue is controller latency is absolutely terrible, but expected for 9600 baudrate. Issue is -- I can't find a way to set baudrate to above 128000? This is with the official driver for this cable. I have yet to try 128000, but currently the latency is completely unusable.

I believe this means it is an issue with this specific cable and it not adjusting to the baudrate of `server.jar`, rather using the driver settings. Does anyone know of a solution to this? Google is getting me nowhere.

Will test 128000 and get back.

javmarina, 2022-04-15:
That looks cool! So after some investigation I think what is happening is that 1M is not a supported baudrate for the CP2102, and when we try setting the baudrate by code (either Python or Java), the device/driver discards it. So it seems we're in fact working with a lower baud rate. That would explain reading such a high number of input bytes.

According to the CP2102/CP2109 [datasheet](https://www.silabs.com/documents/public/data-sheets/CP2102-9.pdf), supported baudrates are (Table 12): 300, 600, 1200, 1800, 2400, 4000, 4800, 7200, 9600, 14400, 16000, 19200, 28800, 38400, 51200, 56000, 57600, 64000, 76800, 115200, 128000, 153600, 230400, 250000, 256000, 460800, 500000, 576000 and 921600.

As you can see, 1M is not there. You might be wondering now why I chose 1M as the default. It's because it is the highest baudrate supported by Arduino Uno - ATmega16u2 (actually the highest is 2M when double speed mode is enabled, but it's very unstable). Coincidentally, it's supported by the FTDI USB-UART adapter (the one I use), so I never had issues.

The issue is then that there is a mismatch between the adapter baudrate and the Arduino baudrate. The criterion for selecting a baudrate is to select values in the intersection of "supported by adapter" (see above for CP210x) and "supported by ATmega16u2", some of which can be seen in Tables 18-9 to 18-12 of the [datasheet](https://ww1.microchip.com/downloads/en/DeviceDoc/doc7799.pdf). They're basically 1M, 0.5M, 250k, 230.4k, 115.2k...

I think the way to go is: first, select a baud rate from the intersection (preferably high). In this case, baud rates supported by **both** ATmega16U2 (Arduino) and CP210x (adapter) are: 0.5M, 250k, 230.4k... (not 1M!!!). So I would go with 0.5M; then, compile the firmware for your AVR device (not limited to Arduino Uno), setting the `BAUD` macro in avr.h to the desired value (a warning will appear if not achievable, look out for it). Then, operate the software as usual: select the baudrate in the Java GUI or Python script, and it should work. Regarding @ethanriverpage's concern of having to use the Device Manager settings, I think it's not necessary as long as we use supported values. The libraries I'm using are capable of reconfiguring the virtual COM driver on the go.

Going forward, the corrective actions are imho:

1. Clearly show this information in the documentation (sort of a guide on how to select the baud rate).
2. In the software side, detect whether the baud rate was set correctly. If not, display a warning to the user

Let me know if you think more actions are needed!

javmarina, 2022-04-15:
I have included some checks in the Python script and the Java GUI, however they are not guaranteed to work because of differences in drivers implementations. Moreover, I have explicitly updated the Java serial library to the latest version and enabled reducing the latency timer for FTDI devices.

Uniersr, 2022-04-16:
I'm so unlucky, i compiled and flashed Joystick.hex with modified BAUD to 115200.
```
python client.py 115200
List of available ports:
[0] COM6
Introduce port index: 0
Writing ['0x0', '0x0', '0x8', '0x80', '0x80', '0x80', '0x80', '0x0', '0x54'] to serial port
Wrote 9 bytes out of 9
Reading 1 byte(s) from serial port
Read 1 bytes: ['0xe0']
[sync()] Not synced, forcing sync...
First, clear input buffer
Then, write nine 0xFF bytes to flush out buffer on device
Writing ['0xff', '0xff', '0xff', '0xff', '0xff', '0xff', '0xff', '0xff', '0xff'] to serial port
Wrote 9 bytes out of 9
11 byte(s) in the input buffer, waiting 18.0 ms for more bytes
18 byte(s) in the input buffer, waiting 18.0 ms for more bytes
23 byte(s) in the input buffer, waiting 18.0 ms for more bytes
30 byte(s) in the input buffer, waiting 18.0 ms for more bytes
35 byte(s) in the input buffer, waiting 18.0 ms for more bytes
42 byte(s) in the input buffer, waiting 18.0 ms for more bytes
47 byte(s) in the input buffer, waiting 18.0 ms for more bytes
52 byte(s) in the input buffer, waiting 18.0 ms for more bytes
59 byte(s) in the input buffer, waiting 18.0 ms for more bytes
65 byte(s) in the input buffer, waiting 18.0 ms for more bytes
70 byte(s) in the input buffer, waiting 18.0 ms for more bytes
76 byte(s) in the input buffer, waiting 18.0 ms for more bytes
82 byte(s) in the input buffer, waiting 18.0 ms for more bytes
86 byte(s) in the input buffer, waiting 18.0 ms for more bytes
92 byte(s) in the input buffer, waiting 18.0 ms for more bytes
98 byte(s) in the input buffer, waiting 18.0 ms for more bytes
104 byte(s) in the input buffer, waiting 18.0 ms for more bytes
111 byte(s) in the input buffer, waiting 18.0 ms for more bytes
116 byte(s) in the input buffer, waiting 18.0 ms for more bytes
123 byte(s) in the input buffer, waiting 18.0 ms for more bytes
128 byte(s) in the input buffer, waiting 18.0 ms for more bytes
134 byte(s) in the input buffer, waiting 18.0 ms for more bytes
145 byte(s) in the input buffer, waiting 18.0 ms for more bytes
149 byte(s) in the input buffer, waiting 18.0 ms for more bytes
Eventually we received 149 byte(s)
[read_byte_latest()] 149 byte(s) in input buffer, reading last one and discarding the rest
Reading 149 byte(s) from serial port
Read 149 bytes: ['0x0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0x0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0xe0', '0x0', '0x0', '0x0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0x0', '0x0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0x0', '0xe0', '0x0', '0x0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0x0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0x0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0x0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0xe0', '0x0', '0x0', '0x0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0x0', '0x0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0x0', '0xe0', '0x0', '0x0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0x0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0x0']
[read_byte_latest()] Input buffer contained: ['0x0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0x0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0xe0', '0x0', '0x0', '0x0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0x0', '0x0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0x0', '0xe0', '0x0', '0x0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0x0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0x0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0x0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0xe0', '0x0', '0x0', '0x0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0x0', '0x0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0x0', '0xe0', '0x0', '0x0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0x0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0x0']
[SYNC ERROR] We expected between 1 and 9 bytes
[sync()] Not synced
Could not sync!
```
Then i adjust the baudrate of the COM6 port in Device Manager, then i disconnected blue cable from nintendo switch to arduino and usb to uart from pc to arduino then i connected first usb to uart to pc from arduino then blue cable to nintendo switch from arduino and nothing worked... I'm so unlucky...
```
python client.py 115200
List of available ports:
[0] COM6
Introduce port index: 0
Writing ['0x0', '0x0', '0x8', '0x80', '0x80', '0x80', '0x80', '0x0', '0x54'] to serial port
Wrote 9 bytes out of 9
Reading 1 byte(s) from serial port
Read 1 bytes: ['0xe0']
[sync()] Not synced, forcing sync...
First, clear input buffer
Then, write nine 0xFF bytes to flush out buffer on device
Writing ['0xff', '0xff', '0xff', '0xff', '0xff', '0xff', '0xff', '0xff', '0xff'] to serial port
Wrote 9 bytes out of 9
[... cut ...]
[read_byte_latest()] Input buffer contained: ['0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0x0', '0x0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0x0', '0xe0', '0x0', '0x0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0x0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0x0']
[SYNC ERROR] We expected between 1 and 9 bytes
[sync()] Not synced
Could not sync!
```
[image]

javmarina, 2022-04-24:
So now we see what you are receving. This is super strange. There is now way that the FW sends those bytes. Are you sure you are correctly flashing the firmware? If so, try to compile the old FW version ([here](https://github.com/javmarina/Nintendo-Switch-Remote-Control/releases/tag/HORIPAD)), because there was a big rewrite afterwards. Maybe I introduced a bug somewhere.

However, I tell you again, it's very strange. Can it be a faulty HW (Arduino, serial adapter, cables...)?

Uniersr, 2022-05-01:
Zip with Joystick.hex and server.jar compiled from https://github.com/javmarina/Nintendo-Switch-Remote-Control/releases/tag/HORIPAD:
[1.zip](https://github.com/javmarina/Nintendo-Switch-Remote-Control/files/8599625/1.zip)
Even modifing band from avr.h from https://github.com/javmarina/Nintendo-Switch-Remote-Control/releases/tag/HORIPAD to 115200 then compiling, It's still a no luck.
Compiling gui from https://github.com/javmarina/Nintendo-Switch-Remote-Control/releases/tag/HORIPAD and running it, leads to
`no main manifest attribute, in server.jar`
Running client.py still nothing.
```
List of available ports:
[0] COM6
Introduce port index: 0
Writing ['0x0', '0x0', '0x8', '0x80', '0x80', '0x80', '0x80', '0x0', '0x54'] to serial port
Wrote 9 bytes out of 9
Reading 1 byte(s) from serial port
Read 1 bytes: ['0xe0']
[sync()] Not synced, forcing sync...
First, clear input buffer
Then, write nine 0xFF bytes to flush out buffer on device
Writing ['0xff', '0xff', '0xff', '0xff', '0xff', '0xff', '0xff', '0xff', '0xff'] to serial port
Wrote 9 bytes out of 9
23 byte(s) in the input buffer, waiting 18.0 ms for more bytes
28 byte(s) in the input buffer, waiting 18.0 ms for more bytes
45 byte(s) in the input buffer, waiting 18.0 ms for more bytes
50 byte(s) in the input buffer, waiting 18.0 ms for more bytes
56 byte(s) in the input buffer, waiting 18.0 ms for more bytes
62 byte(s) in the input buffer, waiting 18.0 ms for more bytes
67 byte(s) in the input buffer, waiting 18.0 ms for more bytes
74 byte(s) in the input buffer, waiting 18.0 ms for more bytes
79 byte(s) in the input buffer, waiting 18.0 ms for more bytes
86 byte(s) in the input buffer, waiting 18.0 ms for more bytes
104 byte(s) in the input buffer, waiting 18.0 ms for more bytes
109 byte(s) in the input buffer, waiting 18.0 ms for more bytes
115 byte(s) in the input buffer, waiting 18.0 ms for more bytes
120 byte(s) in the input buffer, waiting 18.0 ms for more bytes
125 byte(s) in the input buffer, waiting 18.0 ms for more bytes
128 byte(s) in the input buffer, waiting 18.0 ms for more bytes
Eventually we received 128 byte(s)
[read_byte_latest()] 128 byte(s) in input buffer, reading last one and discarding the rest
Reading 128 byte(s) from serial port
Read 128 bytes: ['0x0', '0x0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0x0', '0x0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0x0', '0xe0', '0x0', '0x0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0x0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0x0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0x0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0xe0', '0x0', '0x0', '0x0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0x0', '0x0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0x0', '0xe0', '0x0', '0x0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0x0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0x0']
[read_byte_latest()] Input buffer contained: ['0x0', '0x0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0x0', '0x0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0x0', '0xe0', '0x0', '0x0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0x0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0x0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0x0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0xe0', '0x0', '0x0', '0x0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0x0', '0x0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0x0', '0xe0', '0x0', '0x0', '0xe0', '0x0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0xe0', '0x0', '0x0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0x0', '0xe0', '0xe0', '0x0']
[SYNC ERROR] We expected between 1 and 9 bytes
[sync()] Not synced
Could not sync!
```
Disconnecting everything after this image has been taken then connecting arduino uno r3 to a PC, HORIPAD S is recognized.
[image]
[image]
I think i'm starting to give up until they manage to force avr.h and server.jar or client.py to communicate from RX-TX and TX-RX instead of RX-RX and TX-TX and if will gonna work for me.

Uniersr, 2022-09-18:
@javmarina https://github.com/Kashalls/NintendoSwitchController fixed the issue for me, closing the issue...

https://user-images.githubusercontent.com/99698024/190917297-e28b364c-1be8-475f-a8fd-012c6c4ec26b.mp4

closed 2022-09-18 by Uniersr

Uniersr, 2022-09-18:
The sync is based on luck.

[image]

Uniersr, 2022-09-19:
@javmarina I also figured out the way to fix this issue, by flashing same Joystick.hex compiled with only modified to operate at 115200 baudrate from your Nintendo-Switch-Remote-Control to both atmega328p and atmega16u2, i was able to get it working too!

atmega328p: 
```
cd C:\Program Files (x86)\Arduino\hardware\tools\avr\bin
avrdude -C"C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf" -v -patmega328p -carduino -PCOM3 -b115200 -D -Uflash:w:C:\Users\CENSORED\Downloads1\joyhex\1\Joystick.hex:i

```

atmega16u2: using flip 3.4.7

[image]