javmarina/Nintendo-Switch-Remote-Control #8: Curious about cross-compatibility with my project

nullstalgia, 2022-04-04:
Howdy!

I see we were both inspired by **wchill**'s controller project! I have a similar fork of that meant for easy-to-setup UART->Nintendo Switch connections. Namely, my project [UARTSwitchCon](https://github.com/nullstalgia/UARTSwitchCon). I was wondering if you ever needed Joy-Con support (without IMU/NFC/IR support, but it could be added by someone) you could try out my ESP32-based project! Some games (Like Clubhouse Games) *need* Joy-Cons for the games to work, and I built that for simple games to be played with a Twitch Chat via [ClubchatGames](https://github.com/nullstalgia/ClubchatGames).

It can also act as a Pro Controller, but that's less important when we have the existing projects.

Just wanted to share this and see if it could help you at all! Take care and stay safe.

javmarina, 2022-04-05:
Hi! Thanks for coming here.

I'm currently targetting Arduino Uno (the one I have at home) and AVR devices. For me, changing to a entirely different FW architect would be a lot of work and a waste of time. However, I have looked at your project for inspiration and I think I can make some improvements to mine.

I don't know why I removed the avr_mock, but I'm getting it back.

Also, I'm going to investigate if it's viable to also emulate Joy-Cons. Afaik, there was a Joy-Con grip with a USB connection, and I guess input is sent over USB. I'm limited to one USB interface and no BT connection, so this could work. What fo you think?

nullstalgia, 2022-04-07:
That's an entirely understandable concern with the different architecture. Even I don't fully know how to utilize the esp-idf, I mostly cobbled together an existing wireless controller project with examples concerning UART input then added some of my own logic and various fixes. I'm glad you were able to find some inspiration, though!

I do own the Charging Grip and looked into doing something similar to what you just described but didn't come to any conclusion on whether it would work. I think you have a better chance at figuring that one out, considering you were able to swap the descriptor to the Pro Con instead of the Horipad!

If you need me to get you some information from it (like copying a descriptor with an app, for example), I'll gladly assist however I can.